tomojitakasu/PocketSDR #1: Amazing!

AgileEngineeringLLC, 2022-01-31:
This looks amazing, great work! Is there a way to purchase one of the SDR boards? Don

tomojitakasu, 2022-01-31:
Currently no plan to distribute RF-frontend H/W.
H/W design, F/W and S/W code can be freely available. So you can reproduce it by yourself. However,  at this moment, you may be hard to get MAX2771. All distributers have no stock. The lead time would be over 30 weeks. I expect to improve the situation soon.

AgileEngineeringLLC, 2022-01-31:
Thank you. Really nice work!!

lukejenkins, 2023-12-06:
@AgileEngineeringLLC If you are still interested, there are a few for sale currently on Tindie: https://www.tindie.com/products/kenchitaru/pocketsdr-neo-gnss/

lukejenkins, 2023-12-06:
I've only had mine for a few days, and have only been using the Application Programs by @tomojitakasu so far (thanks for this project!). Combine that with a half dozen hours hacking  around on GNSS-SDR using a hackrf; and I don't see why it wouldn't be a great fit. That is with the usual caveats of either reasonable bandwidth configs for the MAX2771 and/or enough compute power. Hopefully someone else chimes in with first hand experience.

closed 2026-09-01 by tomojitakasu