tomojitakasu/PocketSDR #2: ADC Clock Alignment implementation

Sven-Bauer, 2022-04-08:
I haven't checked the actual PCB layout yet, but based on the current circuit file (pocket_sdr_v2.1_sch.pdf), the ADC clock alignment doesn't seem to be implemented as Maxim intended.

The MAX2771 specifications (19-100378; Rev. 0; 7/18) describe multiple MAX2771s setup in the ADC Clock Alignment section on page 22 as:

> One device would be designated as the clock source and configured to output its ADC clock on its CLKOUT pin. This clock signal can then be buffered and distributed through an external clock tree. The buffered clocks are then input on the ADC_CLKIN pins of **all** MAX2771 devices (including the clock source), and **all** devices are configured to use this external clock as their ADC clock.

In contrast, the PCB schematic shows that the CLKOUT of U1 MAX2771 is connected only to U2 MAX2771, not to itself. Instead, its own ADC_CLKIN is pulled to ground.

tomojitakasu, 2022-04-13:
The current design has no problem but your suggestion may be better for sampling synchronization.

oberstet, 2024-03-10:
first of, awesome project! thanks for publishing as OSS! I'm new to RF but an OSS supporter and contributor as well.

being a RF dummy, may I ask 2 questions? rgd. clocking of PocketSDR - but more in general, so slightly OT wrt to this issue. 

----

The ADI docs of MAX2771 say:

*"The maximum frequency of the pre-divided reference clock is 22MHz if the x2 option is selected, and 11MHz if the x4 option is selected. The ADC sampling clock can then be generated by a second fractional divider."*

PocketSDR is using 24 MHz though (and I can imagine there will be reasons why this is desirable).

How does that work?

----

The MCU _also_ uses 24 MHz, but it is using an own, separate CXO.

Is there a reason of not using e.g. a clock buffer and drive the MCU from the one TCXO as well?

Would there be any advantage of having the clocks of the RF frontend (the PLLs inside) and the MCU running phase synchronized?

In PocketSDR's case, the baseband processing sits beyond USB, and goes through USB profiles, and thus it is not and unlikey could be phase synchronized down to the RF PLL clock, but in general, would there be any advantage of having the RF PLL and baseband processor clocks running phase synchronous (eg driven from the same TCXO)?

tomojitakasu, 2024-03-10:
The TCXO frequency should be changable by replacing it. In fact, I tested 16.368, 24 and 25 MHz for prototypes and seleced 24 MHz.

tomojitakasu, 2024-03-11:
MAX2771 has 3-stage ADC clock generator and only requires MAX 44 MHz as the 3rd divider input. If enabling the first pre-divider, 24 MHz input has no problem Refer Figure 3. 

https://www.analog.com/media/en/technical-documentation/data-sheets/MAX2771.pdf

oberstet, 2024-03-11:
Thanks a lot for commenting! I am trying to reconcile the information with the MAX2771 data sheet, basically understand this stuff (and I'm a newbie, so pls bare with me;)

The MAX2771 data sheet says

* page 9: "Reference Input Frequency: 8 to 44 MHz"
* page 24: "The maximum frequency of the pre-divided reference clock is 22MHz if the x2 option is selected"

You have successfully tested up to 25 MHz.

So that implies PocketSDR is _not_ using pre-division on the reference clock, but 1:1, right?

I don't understand your hint with "3rd divider input" ... the MAX2771 gets one clock input, and then stuff happens inside. But what's the maximum clock (electrically) at its input? Is that 44 MHz?

----

When you say " TCXO frequency should be changable": could I use e.g. 30.72 MHz? I need that one for another RF frontend already. But then, I need 24 MHz for the MCU/MPU I use anyways, so 24 isn't a problem.

> FWIW, I am designing the device with a clock tree I currently plan to look like: https://www.edaboard.com/threads/designing-a-clock-tree-looking-for-integer-n-plls-buffers.410304/

tomojitakasu, 2024-03-11:
I used the pre-divider with 25 MHz XTAL for a prototype. The "3rd divider" means "3rd-stage divider".

The datasheet only says MAX 44 MHz as "Reference Input Frequency".

jmfriedt, 2025-06-17:
I have documented various cases of ADC and LO coherence depending on programming sequence or cold start of the hardware at https://github.com/jmfriedt/max2771_fx2lp/tree/main/250610_calibrate_phase (using the same circuit layout as the 2-channel PocketSDR discussed here). Maybe someone needing phase coherence of the acquisition can find these useful.

closed 2026-09-01 by tomojitakasu