tomojitakasu/PocketSDR #3: Parts selection problem of 3.3V LDO

tomojitakasu, 2022-06-23:
The part for U5 (3.3V LDO) in HW/pocket_sdr_v.2.1.xlsx is not suitable. The output voltage of the LDO may be 5V instead of 3.3V by the current circuit design.

Replace the part for U5 by the proper one, in which the pin 4 of SOT-23-5 package shall be NC (no connection).

osqzss, 2022-06-24:
The V_FB pin (pin 4) of RP111N331D-TR-FE has to be connected to the V_OUT pin (pin 5). You can use a 0603 (1608) 0-ohm SMD resistor  as a quick fix.
[image]

oberstet, 2024-03-10:
The LDO seems to be replaced for *v2.2*, it now is:

LDO 3.3V 500mA SOT23-5 **Nisshinbo RP111N331D-TR-FE**

whereas the *v2.1* used (or at least, the BOM list says):

LDO 3.3V 500mA SOT23-5	**RICOH RP111N331D-TR-FE**

tomojitakasu, 2024-03-10:
RP111N331D-TR-FE is still not suitable, in which pin 4 is not NC. Replace it by another one with pin 4 as NC if using the original PCB design. It will be fixed in the next release.

https://www.digikey.jp/ja/products/detail/nisshinbo-micro-devices-inc/RP111N331D-TR-FE/10244938

closed 2025-04-13 by tomojitakasu