tomojitakasu/PocketSDR #4: The isolation resistor for the RF_DIVIDER is missing.

takiragi, 2022-07-22:
The RF_DIVIDER "PD0922J5050S2HF" requires an external 100 Ohm resistor that is called isolation resistor between two output ports, but that is missing in the circuit.
This issue degrades isolation between the two ports. For example, the reflected signal from LNA-L port interferes with the signal to the LNA-H port.
However, it is not so serious because the return loss on both ports of the MAX2771 is not so high.

tomojitakasu, 2022-07-22:
Thanks for the suggestion.

I found the requirement in the datasheet. It will be added in the next update of the circuit.

https://cdn.ttm.com/repository/products/wireless-xinger/power-dividers/PD0922J5050S2HF/PD0922J5050S2HF_DataSheet(Rev_B).pdf

tomojitakasu, 2025-04-14:
Fixed in Pocket SDR FE 2CH HW v2.3.

https://github.com/tomojitakasu/PocketSDR/tree/master/FE_2CH/HW/v2.3

closed 2025-04-14 by tomojitakasu