tomojitakasu/PocketSDR #12: Extending the code to decode Beidou B2b signal

MayHarryWang, 2024-01-06:
Dear tomojitakasu,

Thanks for sharing the amazing work! 
I have learned a lot of basic SDR knowledges from PocketSDR, and the code is really compact.

Currently, I want to extend the code to decode Beidou B2b signal.
However, I encountered some problems because the LDPC library can only decode binary LDPC code.
The Beidou B2b needs non-binary LDPC decoding. 
According to the ICD, the Extended Min-Sum Method or Fixed Path Decoding Method may required to decode the navigation message.
Is it possible to decode the message with other method or using existing libraries?

Thank you very much! Looking forward to your reply.

Best regards.

tomojitakasu, 2024-01-06:
I have been looking for public library to decode NB-LDPC for B1C, B2a and B2b. Unfortunately , I have not found suitable codes currently.
If you have useful information, it would be helpful.

MayHarryWang, 2024-01-06:
Thanks for your reply.
I will keep searching suitable code. If there's any useful information, I will add comment here.

Thanks!

MayHarryWang, 2024-01-11:
Dear tomojitakasu,

I found an open source code provided by Lab-STICC.
The link is [https://github.com/Lab-STICC-UBS/NB_LDPC_FB_public](url)
The code also applied Extended Min-Sum decoder, however, it looks like simulation not completely a decoder function.
I add this comment for sharing this information.
I'm not sure if it is useful to integrate the library with PocketSDR, but I might try using this code to decode Beidou signal.

Best regards.

MayHarryWang, 2024-01-11:
Dear tomojitakasu,

I found an open source code provided by Lab-STICC.
The link is https://github.com/Lab-STICC-UBS/NB_LDPC_FB_public
The code also applied Extended Min-Sum decoder, however, it looks like simulation not completely a decoder function.
I add this comment for sharing this information.

Best regards.

tomojitakasu, 2024-01-13:
Thanks for helpful information.

Simply to obtain the navigation data, the NB-LDPC decorder can be skipped. That means error-correction is not necessary. Even if the error-correction skipped, the error-detection can be performed by the frame CRC check. Under good C/N0 conditions, the bit errors are rare, and the way may be practical.

In Pocket SDR ver. 0.10, I only implements error-detection without error-correction to decode the navigation data of BDS B1C, B2a, and B2b. Off course, I know error-correction is better, especially under noisy conditions. So I will try the NB-LDPC decorder in the future release.

tomojitakasu, 2024-01-25:
EMS-based NB-LDPC decoder is added  in ver. 0.11 for BDS B1C, B2a and B2b.

MayHarryWang, 2024-01-26:
Dear tomojitakasu,

Thank you very much!
This is the first NB-LDPC decoder I have ever seen in open source SDR for GNSS.

Best regards.

closed 2025-04-13 by tomojitakasu