tomojitakasu/PocketSDR #15: Working with a single MAX2771?

jmfriedt, 2024-05-30:
Since the PocketSDR cannot be readily found assembled, I am trying to see if I can get the proposed code to work with a single MAX2771 as found on the evaluation board
https://www.analog.com/en/resources/evaluation-hardware-and-software/evaluation-boards-kits/max2771evkit.html
which is fitted with a USB->SPI converter but cannot stream IQ data. I have hence additionally acquired an EZ-USB FX2P board 
https://fr.aliexpress.com/item/1005006134347046.html
which I have successfully flashed with the firmware proposed in this repository (PID/VID correct). I have routed the SPI pins (PD0=CS#, PD2=SCLK, PD3=DATA, PD4=lock) to the associated pins of the unpowered microcontroller of the eval board, and routed the IQ stream from the eval board I0,I1,Q0,Q1 to PB0-4 as well as CLK_OUT to IFCLK. Finally SHDN was pulled up to 3V3.

I *think* I can configure the board correctly since 
``sudo ./app/pocket_conf/pocket_conf``
returns a reasonable answer starting with
```
#  Pocket SDR device settings (MAX2771)
#
#  [CH1] F_LO = 1575.420 MHz, F_ADC =  4.000 MHz (IQ), F_FILT =  0.0 MHz, BW_FILT =  2.5 MHz
#  [CH2] F_LO =     -nan MHz, F_ADC = 48.000 MHz (I ), F_FILT =  0.0 MHz, BW_FILT =  2.5 MHz
```
after I ``sudo ./app/pocket_conf/pocket_conf conf/pocket_L1L2_4MHz.conf``.

However
```
sudo ./app/pocket_dump/pocket_dump 
  TIME(s)    T   CH1(Bytes)   T   CH2(Bytes)   RATE(Ks/s)
         0.8   IQ            0   I            0          0.0
```
keeps on returning empty files and 0 Bytes: is there a requirement I am missing for this software to work with a single channel? Am I missing an option that would allow me to collect these data? I have checked that the CLK_OUT->IFCLK is correctly clocked and that the IQ states are changing.

I know this is not strictly PocketSDR related but any hint would be welcome to expand the users of this software (and possibly later assemble a dedicated board).

Thanks.

jmfriedt, 2024-05-31:
One more issue identified: PocketSDR is clocked at 24 MHz, MAX2771EvalBoard is 16.368 MHz. Disconnecting the on-board local oscillator of MAX2771EvalBoard and feeding Ref_In with 24 MHz allows to get the right Clk_Out frequency and some communication over USB, yet still some 
```
libusb bulk transfer error (2)
libusb bulk transfer error (2)
        3.6   IQ     12582912   I      6291456       1734.1
```
error in the transfer.

jmfriedt, 2024-05-31:
The USB configuration sounds good with ``lsusb -v`` display all bulk transfer endpoints
```
Bus 003 Device 094: ID 04b4:1004 Cypress Semiconductor Corp. EZ-USB
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x04b4 Cypress Semiconductor Corp.
  idProduct          0x1004 EZ-USB
  bcdDevice            0.00
  iManufacturer           1 Cypress
  iProduct                2 EZ-USB
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002e
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           4
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x86  EP 6 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x88  EP 8 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled
```
and the IQ stream is fluctuating over time

[image]

but the binary files after complaining about timeout only display 01 values so the USB transfer is failing.

tomojitakasu, 2024-05-31:
To enable bulk-transfer to host by slave-FIFO with external clock, you have to configure the pins of EZ-USB FX2LP:
- RDY0/SLRD pull-up (disabled)
- RDY1/SLWR pull-down (enabled)
- IFCLK/PE0 connected to external clock line.
- PA4/FIFOADR0 pull-down (0)
- PA5/FIFOADR1 pull-up (1)
Please check them.

pocket_conf assumes the TCXO frequency 24 MHz . So wrong frequencies are shown if with other TCXO freqency.

jmfriedt, 2024-06-01:
Thank you for pointing these out. I had missed RDY0 and RDY1 pull up/down indeed, these were missing and I erroneously thought that seeing the bulk transfer endpoints with ``lsusb -v`` was enough to assess proper configuration.
Now ``sudo ./app/pocket_dump/pocket_dump`` is failing with ``libusb_submit_transfer(0) error (-1)`` in the USB event handler thread.

jmfriedt, 2024-06-03:
Is there a way to debug the firmware executed in the FX2LP? My concern is that the firmware is not executing properly, but doesn't the fact that I can read correctly the MAX2771 status after configuring and that the USB interface is configured with the right PID/VID prove that the FX2LP is running correctly? I tested with https://gist.github.com/siddharthdeore/b1aac400ec94ca53730291a92279ea0e and this example is running correctly with bulk endpoint 6 communicating correctly even though I cannot find its FX2LP source code: my concern is that the PocketSDR firmware address range lies between 0x099C and 0x09F8 when the functional example cited above is in the 0000 to 0x0235 so maybe I am doing something wrong with executing the firmware in the 8051 micrcontroller. I flashed using
```
sudo ./fxload -D /dev/bus/usb/003/087 -I PocketSDR/FE_2CH/FW/v2.1/pocket_fw.hex -c 0xc2 -s Vend_Ax.hex -t fx2
```
after unloading the ``usbtest`` kernel module.

Another concern might be that the board I am using is fitted with CY7C68013A-56PVXC whereas PocketSDR is using CY7C68013A-56LTXCT so different pin format: can this make any difference?

tomojitakasu, 2024-06-03:
Some chinese FX2LP EVK boards have wrong silk prints for the header pins. Refer the following blog with Google translate. I also fell into the pitfall and wasted time.

https://blog.goo.ne.jp/osqzss/e/d86df04de96123fd5c73bbb6db6e8bc5

jmfriedt, 2024-06-03:
Yes this was it !
``` 
TIME(s)    T   CH1(Bytes)   T   CH2(Bytes)   RATE(Ks/s)
      3.2   IQ     25690112  IQ     25690112       4025.4
```
Thank you so much for your help. Now I can get to some real work.

jmfriedt, 2024-06-11:
For the record, it all ends up working with a single MAX2771 when
```
sudo ./app/pocket_conf/pocket_conf conf/pocket_L1L6_12MHz.conf 
sudo ./app/pocket_conf/pocket_conf 
sudo ./app/pocket_dump/pocket_dump -r -t 1 ch1.bin ch2.bin             # -r pour ne PAS interleave
python3 ./python/pocket_acq.py ch1.bin -f 12 -fi 3 -sig L1CA -prn 1-32 
```
Explanation:
1. while the pocket_L1L2_4MHz.conf allows for detecting a single tone from a synthesizer (tested at -90 dBm) fed before or after the high-band LNA with a frequency offset of 100 to 500 kHz from L1 carrier, this circuit is far too noisy around baseband to detect GPS. Hence the need to offset the IF by 3 MHz and sample at 12 MS/s
2. verification of the configuration (validates SPI and config)
3. Although I did change in ``src/sdr_conf.c`` (maybe a bad idea)
```
static int max_ch(int type)
{
    switch (type) {
        case TYPE_POCKET_2CH: return 1;
```
by replacing ``return 2`` with ``return 1``, I still had two files created by ``pocket_dump`` => using the ``-r`` switched led to the accumulation of all IQ samples in a single file representing the single MAX2771 I am using
4. Finally acquisition of all the signals as shown below, matching the constellation seen by my mobile phone.

[image]
[image]
[image]

thank you for all the amazing tools !

jmfriedt, 2025-01-30:
Thank you for helping with this project.

closed 2025-01-30 by jmfriedt

smagellan, 2025-08-24:
@jmfriedt What a wonderful setup (no, really!).  
1) I see LNA_LO_IN and MIX_LO_IN are attached to dead-end SMA adapters. Why is that?
2) There is a connection between MIX_HI_IN (J6) and LNA_HI_OUT (J5) via some unknown part. Which part number is that?

jmfriedt, 2025-08-25:
1. I believe I ran out of 50 ohm loads and assembled a couple with 50 ohm SMD resistors on SMA connectors. I must admit I cannot remember the details, but that sounds like the most credible explanation.
2. the band pass filter might not be necessary but I used the MiniCircuits VBF-1575+ I had on my desk to reject out of GNSS L1 band components between J5 and J6.