tomojitakasu/PocketSDR #20: 『Ad 广告』DATAGNSS launched 4CH Pocket SDR

rinex20, 2024-08-18:
DATAGNSS now launched the 4CH version of Pocket SDR. You can find more information here.  

The news page is here:  https://www.datagnss.com/blogs/news/pocketsdr-gnss-receiver-launched 

[image]

rinex20, 2024-08-22:
## Updated 2024/08/22
- The first batch of 5 pcs has been sold out! 
- Pre-orders are now open, with the next shipment expected around September 15th. 
- Orders will be fulfilled in the order they are received! 
- If you are a Chinese user, you can contact our domestic sales channel directly at info@datagnss.com. 

We use a low-cost motherboard, Intel J6412 platform, costing less than $150. It is configured with L1, L2, and L5/E5B frequency bands. The CPU can run at about 60% capacity and temperature is controlled around 65-80 degrees Celsius.

[image]

[image]

rinex20, 2024-08-22:
If you want to track the latest information about the product, please follow this telegram channel:

[t.me/pocketsdr](https://t.me/pocketsdr)

rinex20, 2024-08-23:
## Updated 2024/08/23
The 2nd batch of production for PocketSDR is being prepared, and it's expected that this batch we will produce 5 units (since the last batch produced 10 PCBs), with shipping expected in mid-September. 

Currently, we are accepting pre-orders.

[Pre-order Here](https://www.datagnss.com/blogs/news/pocketsdr-gnss-receiver-launched)

rinex20, 2024-08-26:
[image]

rinex20, 2024-08-28:
## Updated 2024-08-28
The 2nd batch of 5 PocketSDR units has been booked! Currently undergoing SMT production and testing at the factory, estimated shipping time is around **September 10th**.

The 3rd batch of 10 PocketSDR units has started pre-orders, PCB production has begun, estimated shipping by **the end of September**. 

Please place your orders as soon as possible, shipments will be processed in the order they were placed.

[Pre-order](https://www.datagnss.com/collections/evk/products/pocketsdr-gnss-receiver)

datagnss, 2024-09-05:
[image]