tomojitakasu/PocketSDR #21: Which MDK/uVision version is pocket_fw.uv2 created with?

hsm-0510, 2024-08-19:
Hi, I was trying to follow the instructions under the FE_CH2 directory. I tried to open the .uv2 file in UV2 and UV4 both, but the error message remains the same. It can't read the project file
[image]
Can you share with me the MDK version and the u Vision version that was used to develop this project?

tomojitakasu, 2024-08-31:
An older version of EZ-USB FX2LP DVK (cy3684kit_RC8.zip) contains the free and limitted version of Kei uVision2 IDE, which only supports less than 4KB as code size. The F/W for Pocket SDR FE 2CH was build by the Kei uVsion2 IDE.

However, unfortunately, the older DVK package seems discontinued and unavailable in the current formal Infineon site.
Look for the older DVK package in some way, or port the F/W to the newer FX2LP IDE.

hsm-0510, 2024-09-02:
Thanks Professor Takasu, that's a great help. I was actually trying to open the pocket_fw.uV2 file in Keil uVision2, but it kept displaying incorrect MDK version error, but my version matches with yours. Later I found out through Professor Takuji Ebinuma, that the issue was with EOL, and it needed to be converted to Windows CR+LF. I'll connect a link here to the solution: https://blog.goo.ne.jp/osqzss/e/50568db07ff493c5b07e6c0da38c1836

closed 2024-09-02 by hsm-0510