tomojitakasu/PocketSDR #22: Facing register read, write error and bulk transfer timeout when running pocket_conf and pocket_trk

hsm-0510, 2024-09-02:
Hi Professor Takasu, I uploaded the F/W on PocketSDR 2CH using the technique you mentioned in your article: https://gpspp.sakura.ne.jp/diary202109.htm#0923 The process completed successfully, and I got the message "Successfully Programmed" both for Vend_ax.hex and pocket_fw.iic

However, when I connect the PocketSDR and run the pocket_conf command, I get the following register read error and register write error for CH1 and CH2. When I repeatedly run the command the registers that face these errors change, sometimes the error occurs in many registers, other times it only occurs in one register. Only once it showed no error. But the problem still persists. Is the SPI communication between the Cypress controller and MAX2771 the source of this issue? Did you face similar problem?
[image]
When I was trying to run pocket_conf again and again, only once did it show no error. So I ran the pocket_conf -a command to see if the registers were written correctly. After verifying, I ran the pocket_dump command, it ran but no IQ was stored in the binary file. The ch1.bin file created was 0KB, and when I tried to run the pocket_trk command after that it showed no results on the terminal just indication of "bulk transfer timeout".

Is the bulk transfer timeout also because of the SPI communication between Cypress and MAX2771?

tomojitakasu, 2024-09-03:
Try the following procedure.
(1) No need to execute Vend_ax.hex.
(2) Jumper J2 on FE should be OFF to disable EEPROM.
(3) Execute USB Control Center. USB Control Center recognizes FE as "Cypress FX2LP No EEPROM Device".
(4) Jumper J2 should be ON to enable EEPROM.
(5) Select "Cypress FX2LP No EEPROM Device", execute menu Program - FX2 - 64KB EEPROM of USB Control Center, and select FW image pocket_fw.iic to write FW to FE.
(6) After several seconds, you can see "Programming Suceeded" in status bar.
(7) Reset FE by disconnecting and re-connecting USB cable.

hsm-0510, 2024-09-04:
I followed the instructions and saw the same results as before with register read error and bulk transfer timeout in pocket_conf, pocket_dump and pocket_trk. And after dump completion the .bin file size is 0 KB:
[image]
[image]
[image]

tomojitakasu, 2024-09-04:
You don't seem to able to access your FE properly.
(1) Your FE itself does not work well.
(2) Another USB device with VID:PID=04BA:1004 or 04B4:00F1 exists.

hsm-0510, 2024-09-05:
Hi, that's what I thought. When I tested it with a USB2.0 Hub the read register and write register issue got resolved and the configuration settings uploaded successfully to MAX.
[image]
However, when I run the pocket_trk command, I get:
[image]
[image]
[image]
[image]
[image]
There seems to be a libusb bulktransfer error (2) reoccuring. This also happens when i run the pocket_dump:
[image]
[image]
[image]
I also checked the devices connected and their VID:PID below on my system, and multiple 04BA:1004 or 04B4:00F1 VID:PID do not exist in my computer:
[image]

tomojitakasu, 2024-09-05:
(1) Would you show detailed environmet in which did you test. (FE H/W, CPU, OS, version, version of GCC, LIBUSB.)
(2) Test pocket_scan, pocket_conf and pocket_dump first.
(3) After reset your PC and try again.

hsm-0510, 2024-09-05:
1. Environment Details:
- **FE H/W:** Pocket SDR FE_2CH
- **CPU:** WinMate IK32 Intel Core i5-7200U
- **OS/Version:** Ubuntu 22.04.4 LTS (Jammy)
- **GCC Version:**
[image]
- **LIBUSB:**
[image]
2. Tested pocket_scan, pocket_conf and pocket_dump:
[image]
3. Reset the PC but still got the same result libusb bulk transfer error (2)

tomojitakasu, 2024-09-05:
Your FE is recognized only as a USB 1.1 device and FULL speed mode (12Mbps). FE should be operated as USB 2.0 HIGH speed mode (480Mbps).

> (3) BUS= 1 PORT= 2 SPEED=FULL ID=04B4:1004 Cypress EZ-USB

I'm not sure the reason why. The USB host controller may have some limitation to support multiple HIGH speed USB devices. Please disconnect other USB devices or hubs and try again.

hsm-0510, 2024-09-06:
Professor, could the issue be with drivers? I remember installing EZ-USB FX2LP drivers in windows, but I haven't done that for my Linux system. Is this the appropriate link to get the driver for linux?
https://softwaretools.infineon.com/tools/com.ifx.tb.tool.ezusbfx3sdk

tomojitakasu, 2024-09-06:
Driver=[none] by lsusb seems right.
For Linux, Raspberry OS and macOS, Pocket SDR does not use Cypress driver. It uses libusb-1.0 as a generic USB driver instead. So you don't need to install Cypress driver for these OS.
Only for Windows, Pocket SDR uses Cypress driver and API for EZ-USB FX2LP/FX3 (cyusb.sys and CyAPI.a). Earler version Pocket SDR also used libusb-1.0 even for Windows but it had some performance problem (mainly data drop). So I replaced it by Cypress driver.

closed 2025-04-13 by tomojitakasu