tomojitakasu/PocketSDR #36: pocket_trk with raw option?

CESBIOTEAM, 2024-12-04:
hello
you did a wonderful work. I would like in first time to congratulate you.

in second time I would like to have more information concerning -raw option that you have added recently with pocket_trk.
My wish would be to record binary stream in two files (as pocket_dump) but with the possibility to track in live gnss data.

do you think is it possible?
I believe that raw option could be a solution, but I did not understand the format.
with pocket_dump for example I have 8Mo for a file and with -raw option I have 32 Mo...

is it possible to have more information and if it is possible to record AND to track in same time with pocket_trk...
and how to record binary stream channel1 and channel 2 with pocket_trk..

Thankls for your support!

tomojitakasu, 2024-12-05:
The option -raw of pocket_trk outputs IF data in RAW8, RAW16 or RAW32 format, not in INT8 or INT8X2. Refer p.31 of the following slides.

https://gpspp.sakura.ne.jp/paper2005/pocketsdr_seminar_202411_revA.pdf

CESBIOTEAM, 2024-12-05:
Ok it is clear for me !
And just one point. Is there a way to dump or calculate for a specific duration for pocket_trk ?with pocket_dump i notice the -t option but it seems not to be implemented with pocket_trk

And an other point  , is there a way to process tracking with L1 gps on the second Channel (rf2) not for the first one (rf1)? I know with your schematic it Can not work but i change your design to record L1 signal on the second channel also...

tomojitakasu, 2024-12-05:
(1) Use -toff options to start at a specific time. Push Ctr-C to stop tracking.
(2) In FE 2CH, CH1 and CH2 are dedicated to L1 and L2/L5/L6 bands, respectively. In FE 4CH, all CHs can be configured for either L1 or L2/L5/L6 bands.

CESBIOTEAM, 2024-12-07:
Ok I understand. 
And last point concerning the LOG file written by pocket_try. I assume it is possible to plot this file with pocket_plot.py 
anyway I have some problems if for example I write the following command:

[image]

And last point , if I want to plot SNR versus elevation angle (as your document in the last sheet)  , I need to have NAV data. but where are the NAV data. I assume now NAV data are included in LOG file but I am not completely sure

if you can clarify this point, it would be very nice!? thanks so much

tomojitakasu, 2024-12-07:
pocket_plot.py has not been maintained.
It will be removed in the next release.

CESBIOTEAM, 2024-12-07:
Ok yes I saw that the last update was two years ago...And... is there a possibility to use the LOG file ,for exemple  C/NO for a particular PRN versus élévation or Time for exemple ?maybe we may Do that with nmea file recorded but I do not know if a particular library (python or maltab) has already been written to do that...

tomojitakasu, 2025-03-29:
In ver. 0.14, pocket_plot.py is re-written to plot receiver logs generated by pocket_sdr.py or pocket_trk.
The log format is changed in the version. So, it is not compatible with earlier versions and pocket_trk.py.