tomojitakasu/PocketSDR #38: Gps_sdr_sim output is not readable by PocketSDR

cavdar1016, 2024-12-20:
I am generating bin file that contains IQ data in INT8x2 format by using [gps_sdr_sim](https://github.com/osqzss/gps-sdr-sim) . That bin file is readable by some popular SDR projects. But, even I set sample rate, data type and frequency according to my data, PocketSDR can not process it. It can not acquire any signals and not tracking too. I read [your paper](https://gpspp.sakura.ne.jp/paper2005/pocketsdr_seminar_202411_revA.pdf), one of the SDR projects that can read my data is reading data with same format in your paper when specialize data type as INT8x2. What could cause to this problem? Should I preprocess my data before?

osqzss, 2024-12-23:
PocketSDR can accept only 2-bit I/Q values even in INT8x2 format. Please try extra bit-shift to 8-bit ADC option in https://github.com/osqzss/gps-sdr-sim/blob/master/gpssim.c#L2395. 8-bit shift instead of 4 should generate reasonable I/Q values for PocketSDR. This is just a quick workaround and not a proper way to generate 2-bit I/Q samples.

cavdar1016, 2024-12-24:
I did it and I think there is a mismatch between these two systems. Because even I generate data for 90 seconds, PocketSDR is reading data for only 18 seconds when I apply right values. But, if I apply 1/5 of right sampling frequency, Pocket SDR is tracking and finding PVT solutions successfully. I'm not using "-s" option in gps_sdr_sim to use 2.6e6 as sampling frequency but PocketSDR is solving signal on 5.200 MSPS frequency. Could it be related with another option in gps_sdr_sim or this issue is related with PocketSDR?

osqzss, 2025-03-10:
Try multiplying the Q signal by -1 to invert its sign. 
It seems that the sign of the Q signal output from the MAX2771 is inverted.

jmfriedt, 2025-03-27:
May I tentatively state that decoding L1 C/A does not care about I+jQ or I-jQ? Attached is a GNU Radio flowchart in which
1/ a PlutoSDR is generating a L1 C/A signal using ``pluto-gps-sim`` (I am still using the old version I am familiar with). The PlutoSDR is clocked with a high quality OCXO multiplied to reach a good quality 40 MHz reference clock. The L1C/A signal is split and feeds both inputs of the PocketSDR
2/ ``mkfifo /tmp/ch12 && pocket_dump -r /tmp/ch12`` to stream the samples collected from PocketSDR to a 0MQ socket
3/ the attached flowchart reads from the 0MQ socket and converts either to I+jQ or I-jQ depending on the content of the Chunks to Symbols (notice how the bottom is complex conjugate of the top)

[image]

whose original GRC file is at https://github.com/jmfriedt/max2771_fx2lp/blob/main/250124_gnuradio/max2771_12_IQsign.grc

All output files decode correctly the pluto-gps-sim generated signal (correct SV) with ``pocket_acq.py -f 4 -sig L1CA -prn 1-32 /tmp/chaoutc.bin -d 15000``

[image]
[image]

osqzss, 2025-03-28:
The I/Q polarity does not affect signal acquisition since both I + jQ and I - jQ provide the same signal power. However, as seen in your test results, the reversed polarity leads to an opposite sign in Doppler frequencies. As a result, signal tracking will fail.