tomojitakasu/PocketSDR #41: Case B1I, B2I and B3I data cannot be solved by PVT.

niudundepinguo, 2025-01-09:
I downloaded it from the link given. /sample/B1IB2I_20220114_005132_12MHz and . /sample/B1IB3I_20220114_005252_24MHz. attempted PVT settlement using the pocket_trk.c file. However, it was found that only capture and trace can be performed and no PVT settlement result can be generated, is it a problem with the data itself or the parameter settings?

tomojitakasu, 2025-03-26:
In 0.14, PVT is OK only with B1I and B2I.

[image]

tomojitakasu, 2025-03-27:
The link to sample data of FE 8CH (16Msps x 300 s) is added (over 18 GB !). Try it.

https://github.com/tomojitakasu/PocketSDR/blob/master/sample/L1B1G1L2G2L5E5bL6_20250327_0248_16MHz.link

closed 2025-04-14 by tomojitakasu