tomojitakasu/PocketSDR #43: Q sign?

jmfriedt, 2025-03-03:
In ``python/sdr_func.py`` we can see
```Python
    else: # IQ-sampling
        return np.array(raw[0::2] - raw[1::2] * 1j, dtype='complex64')
```
and in ``python/pocket_trk.py`` we have
```Python
    else: # IQ (Q sign inverted in MAX2771)
        buff[ix:ix+N] = np.array(raw[0::2] - raw[1::2] * 1j, dtype='complex64')
```
Where does this sign inversion of the MAX2771 Q v.s I come from? Is it documented? I cannot find in the datasheet. I faced the issue while attempting to decode L5 which needed L5I-jL5Q to correlate.
Thanks

osqzss, 2025-03-10:
I'm experiencing the same issue with the baseband signals generated by GPS-SDR-SIM. 
It doesn't seem to be documented anywhere, but the sign of the Q signal output from the MAX2771 is clearly inverted.

jmfriedt, 2025-03-10:
I am wondering whether it is the MAX2771 or the definition of the relation of I v.s Q in the GNSS signals. Indeed I have collected a GNSS dataset using an AD9361-based RF frontend for reference and am facing a similar issue, namely that GPS L5 correlates with L5I-jL5Q and not with L5I+jL5Q. In the Galileo documentation https://www.gsc-europa.eu/sites/default/files/sites/all/files/Galileo_OS_SIS_ICD_v2.1.pdf I find on page 20 Eq. 3 some -j term and some +j term so I am not clear where the I and Q combination is defined.

tomojitakasu, 2025-03-26:
IF data format in Pocket SDR complies with MAX2771 convention, where Q-sample polarity is inverted from normal definition. Refer to p.13 of pocket_sdr.py help.

https://github.com/tomojitakasu/PocketSDR/blob/master/doc/pocket_sdr_help.pdf

jmfriedt, 2025-03-26:
Thank you for your reply. But where was this information found? In https://www.analog.com/media/en/technical-documentation/data-sheets/MAX2771.pdf page 20 I only see "Two's complement binary" without differentiating I or Q components.  Also I wonder why I have the same issue with data collected from an AD9361.

tomojitakasu, 2025-03-26:
No formal information. Ambiguous in the data-sheet.

I have not yet found any definition of Q-polarity of IF data in normal way. So I am also not confident about "normal defintion" of that.

tomojitakasu, 2025-03-27:
The ION GNSS-Metadata-Standard defines sample representation "IQ", "IQn", "InQ", etc (where 'n' means inversion) for IF data streams in Table 7 . However, there is no defintion about "normal" and "inversion". The document only implies that some RF frontends take opposite sign convention for Q-samples from other RF frontends. If AD9361 uses the same convention as MAX2771, they might be "normal". If you have any information, it would be helpful.

https://github.com/IonMetadataWorkingGroup/GNSS-Metadata-Standard

tomojitakasu, 2025-04-17:
I checked the Q-polarity of the following open IF-data.

(1) L1 raw-IF-data generated by Safran's Skydel simulator (25Msps, 8+8 bits)
https://etsin.fairdata.fi/dataset/63f8b776-680b-4c98-ace7-d5e443f2b1c5

(2) L1 raw-IF-data captured by USRP (4Msps, 16+16 bits)
https://sourceforge.net/projects/gnss-sdr/files/data/2013_04_04_GNSS_SIGNAL_at_CTTC_SPAIN.tar.gz

Both Q-polaties are opposite to MAX2771.

closed 2026-09-01 by tomojitakasu