tomojitakasu/PocketSDR #46: Latency of the Solution

tevfikozgu, 2025-04-02:
Initially thanks for this great project. As I see, in the new version you added latency of the current solution in ms precision. Is there any other value that gives better resolution? And also what is the meaning of TOFF_L1CA and other defines? Why did you add these values to TOW of ephemeris?

Thanks

tomojitakasu, 2025-04-03:
(1) In usual OS including Windows and Linux, the processing timing can be controled only with the resolution of 10 ms. So, under 10 ms of the latency might be meaningless.

(2) In the current Pocket SDR implementation, the synchronization of the navigation data is done by using the preamble of the current subframe and one of the next subframe. It brings a little delay. So, TOFF_L1CA = 0.16 s = 20 ms x 8 (the number of the preamble bits) is used to correct the delay to obtain proper TOW.

tevfikozgu, 2025-04-04:
First of all, thank you very much for your response. I noticed that while **setting the time (pvt->time)** and **index (pvt->ix)** of the first PVT solution, the values are derived from the ch pointer. In this case, the first PVT time is calculated by adding _sdr_epoch_ to the tow value of the first channel found. However, when determining the index, **0.07 seconds (which I assume is the average propagation delay) is subtracted**. Doesn't this create an inconsistency between pvt->time and pvt->ix? Wouldn't it be **more accurate to also subtract 0.07 seconds from pvt->time as well?**