tomojitakasu/PocketSDR #48: Issue with Time Sync and PVT Solution Using GLONASS Constellation on Pocket SDR

MuhammadQaisarAli, 2025-04-17:
**Description:** I am attempting to use only the GLONASS constellation for PVT (Position, Velocity, Time) solutions, but I am facing a problem where, even after locking onto 8 satellites within 8 seconds, I am unable to achieve time synchronization or a PVT solution. This issue persists for over 8 minutes. I am using a Pocket SDR with 4 channels.

**Steps to Reproduce:**

- Use Pocket SDR (4 channels) to receive signals.
- Attempt to lock onto GLONASS satellites.
- Wait for time synchronization and PVT solution.
- Observe the issue where no time sync or PVT solution is achieved after 8 minutes despite 8 satellites being locked within 8 seconds.

**Expected Behavior:**
 
- After locking onto 8 satellites, I should receive a time sync and PVT solution.
- The GLONASS constellation should be sufficient to provide a working PVT solution.

**Actual Behavior:**

- After 8 seconds, 8 satellites are locked, but no time sync or PVT solution is obtained.
- This issue persists for more than 8 minutes.

**Environment:**

- Pocket SDR (4 channels)
- GLONASS constellation only

**Possible Causes:**

- Configuration issue with Pocket SDR or software.
- GLONASS-specific signal processing issue.
- Potential synchronization issue with the GLONASS signals.

**Additional Information:**
Images can be found below please.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

tomojitakasu, 2025-04-17:
In ver.0.14, GLONASS-only-PVT is not yet supported.

It is a little hard to resolve date-ambiguity in GLONASS pseudorange without external date information.

Please add GPS or Galileo signals tracked togerther.

MuhammadQaisarAli, 2025-04-17:
Thanks professor for prompt response.  
Is the issue of Glonass is specific to pocket sdr only, or other commercial receiver are unable to calculate PVT on Glonass only?.
I think in that case only a single satellite from other constellation will be enough, just to give date or time. 
Moreover, what I found is that the SDR doesn't store or read ephermisis data, which makes makes the SDR acquisition time very long. If we can store and load the ephermisis data, it will just search for available SATs instead of all possible channels.

MuhammadQaisarAli, 2025-04-17:
And if problem is only to due to date, we can't we feed initial date from computer (computing device) which is mostly accurate at least to a day.

tomojitakasu, 2025-04-17:
(1) I'm not sure about commercial receivers. It's rare to operate them withtout GPS.
(2) The current version does not support warm-start with almanac. I'm considering to add it as an option in next release to shorten signal acquisition time.
(3) It's fine for real-time-mode but not suitable for replay-mode.

MuhammadQaisarAli, 2025-04-17:
Thanks for the insights professor. Hope the improvements comes soon.