tomojitakasu/PocketSDR #49: Title: Discrepancy in Galileo E6 Bandwidth Between Documentation and ESA Data

MuhammadQaisarAli, 2025-04-18:
**Description:**
Dear professor, I have noticed a discrepancy between the bandwidths for the Galileo E6 signal in the documentation provided for PocketSDR and the data available on the European Space Agency (ESA) website.

- **PocketSDR Documentation:** According to the signal bands chart in the [PocketSDR documentation](https://github.com/tomojitakasu/PocketSDR/blob/master/doc/signal_bands.pdf), the Galileo E6 signal has a bandwidth of approximately 20 MHz.

- **European Space Agency (ESA):** However, the [ESA Galileo Signal Plan](https://gssc.esa.int/navipedia/index.php/Galileo_Signal_Plan) lists the bandwidth for Galileo E6 as 40.92 MHz.lileo E6 signal has a bandwidth of approximately 20 MHz.

[image]

[image]

This discrepancy could lead to issues in configuration and signal processing, so clarification or an update might be necessary. Could the documentation be updated to reflect the correct bandwidth based on the official ESA data?

tomojitakasu, 2025-04-18:
The modulation of Galileo E6B or E6C is BPSK(5), which has the null-to-null BW (main-lobe signal BW) of approx 10 MHz (not 20 MHz). Most of the signal power is in the BW. So usually no need to track signals outside of the BW.
 
I'm not sure the strict defintion of "Receiver Reference Bandwidth" in Galileo OS SIS ICD but the defintion of the BW may be different.
(It probably includes side-lobe signals.)

tomojitakasu, 2025-04-18:
Sorry, E6 in Galileo may include a restricted signal E6A, which has the modulation of BOC(10,5).

Pocket SDR only supports open signals, which is E6B or E6C for Galileo. So only open signals are shown in the signal-band chart.

MuhammadQaisarAli, 2025-04-18:
Thank you Professor for clarification, indeed indepth knowledge. Although i compiled the signal bands center frequencies and bandwidths from the PDF of bands provided by you, but some I had to take from internet source (ESA). Can you please update the follwoing dictonary as per your bands bandwidths? It will help me to create a plugin for pocket sdr to automatically create config file for user selected gnss bands.

```
SIGNAL_DEFS = {
        "GPS": {
            "range": "1-32",
            "signals": {
                "L1CA": {"freq": 1575.42, "bw": 2.046, "band": (1575.42-1.023, 1575.42+1.023)},
                "L1CD": {"freq": 1575.42, "bw": 2.046, "band": (1575.42-1.023, 1575.42+1.023)},
                "L1CP": {"freq": 1575.42, "bw": 2.046, "band": (1575.42-1.023, 1575.42+1.023)},
                "L2CM": {"freq": 1227.60, "bw": 2.046, "band": (1227.60-1.023, 1227.60+1.023)},
                "L2C": {"freq": 1227.60, "bw": 2.046, "band": (1227.60-1.023, 1227.60+1.023)},
                "L5I": {"freq": 1176.45, "bw": 20.46, "band": (1176.45-10.23, 1176.45+10.23)},
                "L5Q": {"freq": 1176.45, "bw": 20.46, "band": (1176.45-10.23, 1176.45+10.23)}
            }
        },
        "GLONASS": {
            "range": "-7-6/1-27",
            "signals": {
                "G1CA": {"freq": 1602.00, "bw": 8.0, "band": (1602.00-4.0, 1602.00+4.0)},
                "G1CD": {"freq": 1602.00, "bw": 8.0, "band": (1602.00-4.0, 1602.00+4.0)},
                "G1CP": {"freq": 1602.00, "bw": 8.0, "band": (1602.00-4.0, 1602.00+4.0)},
                "G2CA": {"freq": 1246.00, "bw": 8.0, "band": (1246.00-4.0, 1246.00+4.0)},
                "G2CP": {"freq": 1246.00, "bw": 8.0, "band": (1246.00-4.0, 1246.00+4.0)},
                "G3CD": {"freq": 1202.025, "bw": 8.0, "band": (1202.025-4.0, 1202.025+4.0)},
                "G3CP": {"freq": 1202.025, "bw": 8.0, "band": (1202.025-4.0, 1202.025+4.0)}
            }
        },
        "Galileo": {
            "range": "1-36",
            "signals": {
                "E1B": {"freq": 1575.42, "bw": 4.092, "band": (1575.42-2.046, 1575.42+2.046)},
                "E1C": {"freq": 1575.42, "bw": 4.092, "band": (1575.42-2.046, 1575.42+2.046)},
                "E5AI": {"freq": 1176.45, "bw": 20.46, "band": (1176.45-10.23, 1176.45+10.23)},
                "E5AQ": {"freq": 1176.45, "bw": 20.46, "band": (1176.45-10.23, 1176.45+10.23)},
                "E5BI": {"freq": 1207.14, "bw": 20.46, "band": (1207.14-10.23, 1207.14+10.23)},
                "E5BQ": {"freq": 1207.14, "bw": 20.46, "band": (1207.14-10.23, 1207.14+10.23)},
                "E6B": {"freq": 1278.75, "bw": 30.69, "band": (1278.75-15.345, 1278.75+15.345)},
                "E6C": {"freq": 1278.75, "bw": 30.69, "band": (1278.75-15.345, 1278.75+15.345)}
            }
        },
        "QZSS": {
            "range": "1-9",
            "signals": {
                "L1CA": {"freq": 1575.42, "bw": 2.046, "band": (1575.42-1.023, 1575.42+1.023)},
                "L1CB": {"freq": 1575.42, "bw": 2.046, "band": (1575.42-1.023, 1575.42+1.023)},
                "L1CD": {"freq": 1575.42, "bw": 2.046, "band": (1575.42-1.023, 1575.42+1.023)},
                "L1CP": {"freq": 1575.42, "bw": 2.046, "band": (1575.42-1.023, 1575.42+1.023)},
                "L2CM": {"freq": 1227.60, "bw": 2.046, "band": (1227.60-1.023, 1227.60+1.023)},
                "L5I": {"freq": 1176.45, "bw": 20.46, "band": (1176.45-10.23, 1176.45+10.23)},
                "L5Q": {"freq": 1176.45, "bw": 20.46, "band": (1176.45-10.23, 1176.45+10.23)},
                "L6D": {"freq": 1278.75, "bw": 30.69, "band": (1278.75-15.345, 1278.75+15.345)},
                "L6E": {"freq": 1278.75, "bw": 30.69, "band": (1278.75-15.345, 1278.75+15.345)}
            }
        },
        "BeiDou": {
            "range": "1-63",
            "signals": {
                "B1I": {"freq": 1561.098, "bw": 4.092, "band": (1561.098-2.046, 1561.098+2.046)},
                "B1CD": {"freq": 1575.42, "bw": 4.092, "band": (1575.42-2.046, 1575.42+2.046)},
                "B1CP": {"freq": 1575.42, "bw": 4.092, "band": (1575.42-2.046, 1575.42+2.046)},
                "B2AD": {"freq": 1176.45, "bw": 20.46, "band": (1176.45-10.23, 1176.45+10.23)},
                "B2AP": {"freq": 1176.45, "bw": 20.46, "band": (1176.45-10.23, 1176.45+10.23)},
                "B2I": {"freq": 1207.14, "bw": 4.092, "band": (1207.14-2.046, 1207.14+2.046)},
                "B2BI": {"freq": 1207.14, "bw": 20.46, "band": (1207.14-10.23, 1207.14+10.23)},
                "B3I": {"freq": 1268.52, "bw": 20.46, "band": (1268.52-10.23, 1268.52+10.23)}
            }
        },
        "NavIC": {
            "range": "1-14",
            "signals": {
                "L5SPS": {"freq": 1176.45, "bw": 24.0, "band": (1176.45-12.0, 1176.45+12.0)},
                "SSPS": {"freq": 2492.028, "bw": 16.5, "band": (2492.028-8.25, 2492.028+8.25)},
                "L1SP": {"freq": 1575.42, "bw": 2.046, "band": (1575.42-1.023, 1575.42+1.023)},
                "L5P": {"freq": 1176.45, "bw": 24.0, "band": (1176.45-12.0, 1176.45+12.0)},
                "S": {"freq": 2492.028, "bw": 16.5, "band": (2492.028-8.25, 2492.028+8.25)}
            }
        },
        "SBAS": {
            "range": "120-158",
            "signals": {
                "L1CA": {"freq": 1575.42, "bw": 2.046, "band": (1575.42-1.023, 1575.42+1.023)},
                "L5I": {"freq": 1176.45, "bw": 24.0, "band": (1176.45-12.0, 1176.45+12.0)},
                "L5Q": {"freq": 1176.45, "bw": 24.0, "band": (1176.45-12.0, 1176.45+12.0)}
            }
        }
    }
```

tomojitakasu, 2025-04-20:
Bandwidth
- BPSK(k): 2.048 * k MHz
- BOC(k,k): 4.096 * k MHz
- MBOC(m,k,q): 4.096 * k MHz (only (1-q) power)

https://github.com/tomojitakasu/PocketSDR/blob/master/doc/signal_IDs.pdf

MuhammadQaisarAli, 2025-04-21:
Thanks very much professor.

closed 2026-09-01 by tomojitakasu