tomojitakasu/PocketSDR #52: Potential License Conflict: BSD 2-Clause Project Using GPLv2 Library (FFTW3)

6661661161, 2025-05-05:
## Summary

PocketSDR is currently licensed under the BSD 2-Clause license (`LICENSE.txt`). However, it utilizes the FFTW3 library, which is licensed under GPLv2. This dependency might require PocketSDR itself to be licensed under GPLv2 due to the terms of the GPL.

## Details

PocketSDR uses FFTW3 in the following files:
*   `src/sdr_code.c`
*   `src/sdr_func.c`
*   `python/fftw_wisdom.py`
*   `app/pocket_trk/fftw_wisdom.c`

FFTW3 is licensed under GPLv2: [https://www.fftw.org/doc/License-and-Copyright.html](https://www.fftw.org/doc/License-and-Copyright.html)
The GPLv2 license text can be found here: [https://www.gnu.org/licenses/old-licenses/gpl-2.0.en.html](https://www.gnu.org/licenses/old-licenses/gpl-2.0.en.html)

### GPLv2 Requirements

GPLv2 defines 'a "work based on the Program"' as:
> either the Program or any derivative work under copyright law: that is to say, a work containing the Program or a portion of it, either verbatim or with modifications and/or translated into another language. (Section 0)

Source files linking against FFTW3 likely fall under this definition. This interpretation is supported by:
*   The LGPLv2.1 preamble ([https://www.gnu.org/licenses/old-licenses/lgpl-2.1.en.html](https://www.gnu.org/licenses/old-licenses/lgpl-2.1.en.html)), which states:
    > When a program is linked with a library, whether statically or using a shared library, the combination of the two is legally speaking a combined work, a derivative of the original library. The ordinary General Public License therefore permits such linking only if the entire combination fits its criteria of freedom.
    (Note: `libfec` used in `python/sdr_fec.py` is LGPLv2.1).
*   The GPLv3 FAQ ([https://www.gnu.org/licenses/gpl-faq.html#GPLStaticVsDynamic](https://www.gnu.org/licenses/gpl-faq.html#GPLStaticVsDynamic)), which states:
    > "Linking a GPL covered work statically or dynamically with other modules is making a combined work based on the GPL covered work."

Furthermore, GPLv2 requires that such "works based on the Program" must be licensed under GPLv2:
> You must cause any work that you distribute or publish, that in whole or in part contains or is derived from the Program or any part thereof, to be licensed as a whole at no charge to all third parties under the terms of this License. (Section 2.b)

Section 2 also clarifies that this requirement applies to the work as a whole:
> These requirements apply to the modified work as a whole. [...] But when you distribute the same sections as part of a whole which is a work based on the Program, the distribution of the whole must be on the terms of this License, whose permissions for other licensees extend to the entire whole, and thus to each and every part regardless of who wrote it.

Therefore, it seems likely that the entire PocketSDR project may need to be licensed under GPLv2.

## Possible Solutions

1.  **Change PocketSDR's license to GPLv2.**
2.  **Obtain a commercial license for FFTW3.** FFTW3 offers commercial licenses for a fee: [https://tlo.mit.edu/industry-entrepreneurs/available-technologies/fftw-fastest-fourier-transform-west](https://tlo.mit.edu/industry-entrepreneurs/available-technologies/fftw-fastest-fourier-transform-west)
3.  **Replace FFTW3 with an alternative FFT library.** Benchmarks for alternative libraries can be found here: [https://github.com/project-gemmi/benchmarking-fft](https://github.com/project-gemmi/benchmarking-fft)# Potential License Conflict: BSD 2-Clause Project Using GPLv2 Library (FFTW3)

## Summary

PocketSDR is currently licensed under the BSD 2-Clause license (`LICENSE.txt`). However, it utilizes the FFTW3 library, which is licensed under GPLv2. This dependency might require PocketSDR itself to be licensed under GPLv2 due to the terms of the GPL.

## Details

PocketSDR uses FFTW3 in the following files:
*   `src/sdr_code.c`
*   `src/sdr_func.c`
*   `python/fftw_wisdom.py`
*   `app/pocket_trk/fftw_wisdom.c`

FFTW3 is licensed under GPLv2: [https://www.fftw.org/doc/License-and-Copyright.html](https://www.fftw.org/doc/License-and-Copyright.html)
The GPLv2 license text can be found here: [https://www.gnu.org/licenses/old-licenses/gpl-2.0.en.html](https://www.gnu.org/licenses/old-licenses/gpl-2.0.en.html)

### GPLv2 Requirements

GPLv2 defines 'a "work based on the Program"' as:
> either the Program or any derivative work under copyright law: that is to say, a work containing the Program or a portion of it, either verbatim or with modifications and/or translated into another language. (Section 0)

Source files linking against FFTW3 likely fall under this definition. This interpretation is supported by:
*   The LGPLv2.1 preamble ([https://www.gnu.org/licenses/old-licenses/lgpl-2.1.en.html](https://www.gnu.org/licenses/old-licenses/lgpl-2.1.en.html)), which states:
    > When a program is linked with a library, whether statically or using a shared library, the combination of the two is legally speaking a combined work, a derivative of the original library. The ordinary General Public License therefore permits such linking only if the entire combination fits its criteria of freedom.
    (Note: `libfec` used in `python/sdr_fec.py` is LGPLv2.1).
*   The GPLv3 FAQ ([https://www.gnu.org/licenses/gpl-faq.html#GPLStaticVsDynamic](https://www.gnu.org/licenses/gpl-faq.html#GPLStaticVsDynamic)), which states:
    > "Linking a GPL covered work statically or dynamically with other modules is making a combined work based on the GPL covered work."

Furthermore, GPLv2 requires that such "works based on the Program" must be licensed under GPLv2:
> You must cause any work that you distribute or publish, that in whole or in part contains or is derived from the Program or any part thereof, to be licensed as a whole at no charge to all third parties under the terms of this License. (Section 2.b)

Section 2 also clarifies that this requirement applies to the work as a whole:
> These requirements apply to the modified work as a whole. [...] But when you distribute the same sections as part of a whole which is a work based on the Program, the distribution of the whole must be on the terms of this License, whose permissions for other licensees extend to the entire whole, and thus to each and every part regardless of who wrote it.

Therefore, it seems likely that the entire PocketSDR project may need to be licensed under GPLv2.

## Possible Solutions

1.  **Change PocketSDR's license to GPLv2.**
2.  **Obtain a commercial license for FFTW3.** FFTW3 offers commercial licenses for a fee: [https://tlo.mit.edu/industry-entrepreneurs/available-technologies/fftw-fastest-fourier-transform-west](https://tlo.mit.edu/industry-entrepreneurs/available-technologies/fftw-fastest-fourier-transform-west)
3.  **Replace FFTW3 with an alternative FFT library.** Benchmarks for alternative libraries can be found here: [https://github.com/project-gemmi/benchmarking-fft](https://github.com/project-gemmi/benchmarking-fft)

tomojitakasu, 2025-05-07:
Thanks.

I’ve been looking into similar issues and discussions regarding the FFTW license conflict, with some help from ChatGPT.
Here are some relevant references I found:

- Bob project: https://gitlab.idiap.ch/bob/bob/-/issues/176
- xtensor-fftw: https://github.com/xtensor-stack/xtensor-fftw/issues/36
- SHTOOLS: https://shtools.github.io/SHTOOLS/fortran-faq.html
- Julia FFTW.jl: https://github.com/JuliaMath/FFTW.jl/blob/master/README.md
- pyFFTW: https://github.com/pyFFTW/pyFFTW/issues/229
- liquid-dsp: https://github.com/jgaeddert/liquid-dsp/issues/322

If any other suggestions or related cases, they would be very helpful.

6661661161, 2025-05-08:
Thank you for your response.

In my original post #52, I noted that PocketSDR utilizes the FFTW3 library, which is licensed under GPLv2. This was based on the fact that the current build instructions effectively lead to using FFTW from the OS or MinGW repositories, which typically provide the GPLv2 version.

The examples you provided were very helpful. Based on them, I realized that if the method of obtaining FFTW is explicitly left to the user, rather than implicitly relying on system-provided packages, it is possible to avoid the GPLv2 licensing implication. This leads to an alternative approach that may help preserve the current BSD 2-Clause license of PocketSDR.

Specifically, the following measures could be considered:

- Clarify in the build instructions that FFTW is available under both GPLv2 and a commercial license. Users should be informed that if they choose the GPLv2 version, their build of PocketSDR will be subject to the terms of that license.

- Include system-provided FFTW (e.g., from OS or MinGW repositories) as one option, but explicitly state that such packages are typically GPLv2-licensed and will impose corresponding obligations on the resulting binary.

- Discontinue distribution of binaries that are likely linked with the GPLv2 version of FFTW. Instead, provide source-only releases so that users can build the software in accordance with their own licensing needs.

I hope this suggestion is helpful.

closed 2026-05-12 by hal1278