tomojitakasu/PocketSDR #57: L2, L5, B2I, B3I signals cannot be solved by PVT

niudundepinguo, 2025-07-17:
I downloaded /sample/L1B1G1L2G2L5E5bL6_20250327_0248_16MHz.bin from the given link, and tried to use pocket_trk.c file for PVT settlement. In BDS and GPS systems, I found that only B1I and L1CA signals can be captured, tracked, and PVT solved, but other signals can only be captured and tracked, and PVT settlement results cannot be generated. Does Pocket SDR not support PVT solution for signals other than B1I and L1CA signals? Which signals in BDS and GPS systems does version 0.14 support PVT solution?

tomojitakasu, 2025-07-17:
Pocket SDR calls just the RTKLIB single frequency SPP routine, which only supports L1 signals.
Pocket SDR also utilizes the RTKLIB navigation data decoder, which only supports GPS LNAV, GLONASS L1C/A or L2C/A nav, Galileo I/NAV or F/NAV, BDS D1 or D2 nav, NavIC L5-SPS nav. Other navigation data are only captured as raw navigation bits, but not used for PVT. 
So, for PVT, you need to track GPS L1C/A (L1CA), GLONASS L1C/A (G1CA), Galileo E1B (E1B), QZSS L1C/A (L1CA) or/and BDS B1I (B1I).

niudundepinguo, 2025-07-22:
Thank you very much. I have another question. If I modify the PVT-related code in Pocket SDR, such as the interface parameters of RTKLIB, can I use some signals in GPS L2, L5 or BDS B2I, B3I for PVT solution?

tomojitakasu, 2025-07-30:
All the code is open. You can do anything.

closed 2026-09-01 by tomojitakasu