tomojitakasu/PocketSDR #59: Question about 8CH PocketSDR for beamforming algorithm testing

albs975, 2025-07-24:
Hello,

I’m interested in using the Pocket SDR FE 8CH (with 8 RF inputs) as a test hardware platform for beamforming algorithm implementation.

Is this 8-channel version suitable for real-time beamforming tests (e.g., GNSS or wideband signal processing)?

Is it possible to obtain or order the 8-input/8-channel hardware version?

Thank you!

tomojitakasu, 2025-07-30:
(1) Yes, that was the intention. I designed the FE 8CH primarily for antenna array applications, but a few issues still remain.
The first issue is phase coherency between the RF channels. The FE 8CH employs eight MAX2771, all driven by a common reference clock. However, it is difficult to synchronize the lock timing of each individual PLL within MAX2771. This means the inter channel biases (ICBs) change every time the device is powered on, even if once precisely calibrated.
The second issue is 2- or 3-bit ADC of MAX2771. This provides insufficient dynamic range for anti-jamming applications, which typically require 12- or 14-bit ADC. However, 2- or 3-bit ADC may be adequate for multi-path mitigation.

(2) Not at the moment. I have heard that DataGNSS is planning to produce a clone of FE 8CH after releasing FE8CH-2ANT version.
https://www.datagnss.jp/collections/evk/products/pocketsdr-gnss-receiver?shpxid=428423b7-afe5-49ee-b575-c9325bbdc41b

albs975, 2025-08-09:
Hi professor,

Thanks for the detailed reply about the FE 8CH.

Two follow-ups:

RTL-SDR backend: Is it possible to run the Pocket SDR software stack with RTL-SDR dongles (via librtlsdr) for testing? 

If yes, can the software be configured for a multi-dongle RTL-SDR setup (KrakenSDR etc.) to enable coherent beamforming experiments? 

Thanks!

albs975, 2025-08-28:
Hi Prof. Takasu,

Thank you for your earlier clarification about the FE 8CH and the limitations arising from (1) PLL lock timing/ICB variation across MAX2771s and (2) the on-chip 2–3-bit ADC dynamic range.

I’d like to confirm a possible approach for real-time antenna-array beamforming:

External high-resolution ADC path

If we tap the MAX2771 analog IF/baseband outputs and sample them with an external 12–14 bit ADC (one per channel), do you see any blockers to performing robust array processing (DoA/beamforming/nulling) on the FE 8CH?

In other words, in strong interference, the composite signal amplitude at the array elements can vary widely. Can the MAX2771’s internal AGC handle this without introducing channel-dependent gain/phase transients that would break array coherence?

As a side note about this, I noticed in configuration files in the PocketSDR repo such as

> GAINREF = 60  # AGC gain reference value (0-4095)

What exactly should I understand from this setting in terms of AGC behavior across channels?

If you can share constraints or a known-good configuration (AGC settings, gain values, IF bandwidth) for coherent multi-channel capture, that would be immensely helpful.

Thanks again for your time and for making FE 8CH available for array experimentation!

closed 2026-09-01 by tomojitakasu